pftf/RPi3 #3: Network Booting UEFI for Network Booting Kernel

gordan-bobic, 2019-10-18:
Thanks for an awesomely useful project to give the Pi a standards compliant boot loader.

I have UEFI booting via DHCP+TFTP without a local storage device on the Pi. This works great, but I would like to, from there, TFTP download and boot the dtb/kernel/inirtd. This leads me to two issues:

1. There appears to be no TFTP command in the shell. Is there a way to gain that?

2. Once I can get past 1), I still have the issue of instructing the TFTP booted Tianocore to "do stuff", either
2.1) via the configuration menu which has no local storage to save the NVRAM to (does it?)
2.2) via sending a custom startup.nsh via TFTP

How do I go about pushing out a custom startup.nsh script that makes use of TFTP (or httpboot) to fetch and boot a dtb/kernel/initrd?

pbatard, 2019-10-18:
> 1. There appears to be no TFTP command in the shell. Is there a way to gain that?

There is: https://github.com/tianocore/edk2/tree/master/ShellPkg/DynamicCommand/TftpDynamicCommand

I guess I'll need to send a patch to edk2-platforms so that we add TFTP support to the RPi3 Shell. Then I'll publish a new firmware that includes the command here.

> 2. Once I can get past 1), I still have the issue of instructing the TFTP booted Tianocore to "do stuff", either
2.1) via the configuration menu which has no local storage to save the NVRAM to (does it?)

Currently the 'NVRAM' is the `RPI-EFI.fd` file itself. In other words, there's a dedicated area of the file that acts as NVRAM, and that you can use as such. So if needed, we can add network configuration options there, and they will be preserved between reboots (but not when "upgrading" the firmware).

> 2.2) via sending a custom startup.nsh via TFTP

No idea if/how that would work. You may want to have a look at what other platforms are doing when it comes to TFTP boot.

pbatard, 2019-10-18:
That went faster than expected.

TFTP support for the Pi 3 has now been added in mainline, so, once firmware 1.9 has been built and uploaded (this is a semi-automated process, so, provided everything goes well, it will take a few minutes), you should be able to use that command.

This issue will be closed as a result, as I consider that the other questions are up to you to look into at this stage, and then let me know if you think the firmware needs to be altered further, and if so, how.

closed 2019-10-18 by pbatard

kailiu42, 2020-05-06:
> Currently the 'NVRAM' is the `RPI-EFI.fd` file itself. In other words, there's a dedicated area of the file that acts as NVRAM, and that you can use as such. So if needed, we can add network [...]

Is there a way to tell the offset and size of the block for NVRAM storage in the file? I would like to preserve my NVRAM when upgrading the firmware.

commit 97a94e6e (2019-10-18): Enable TFTP command in UEFI Shell

* Closes #3