pftf/RPi3 #5: openssl/e_os.h is not found in packages path

jiez, 2020-02-11:
I cloned and tried to build it using build_firmware.sh. But I got this error

Processing meta-data .
Architecture(s)  = AARCH64
Build target     = DEBUG
Toolchain        = GCC5

Active Platform          = /home/jie/sources/rpi3-firmware/edk2-platforms/Platform/RaspberryPi/RPi3/RPi3.dsc
..

build.py...
/home/jie/sources/rpi3-firmware/edk2/CryptoPkg/Library/OpensslLib/OpensslLib.inf(-1): error 000E: File/directory not found in workspace
	openssl/e_os.h is not found in packages path:
	/home/jie/sources/rpi3-firmware/edk2
	/home/jie/sources/rpi3-firmware/edk2-platforms
	/home/jie/sources/rpi3-firmware/edk2-non-osi
 

- Failed -
Build end time: 09:33:02, Feb.11 2020
Build total time: 00:00:03

pbatard, 2020-02-11:
You need to initialize the submodules when cloning the git directories. If you need help on this google "git submodule init and update"

closed 2020-02-11 by pbatard

jiez, 2020-02-11:
I did git submodule init and update.

pbatard, 2020-02-11:
Then it should work.

Sorry, but your issue is clearly environmental here and is not something you should ask us to look into. As long as the AppVeyor build fails in the same way as yours (which it doesn't), I'm afraid that this is something you're just going to have to figure out on your own because there's only so much help we can provide on a project that is devoted only to production of binaries for a specific case (AppVeyor automated builds) and is not meant to be used as a starting point for building edk2-platforms.