pftf/RPi3 #6: custom dtb file not working for Debian arm64 buster

jiez, 2020-02-11:
I followed the instructions on https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html . I first did it for Debian arm64 testing (bullseye). I modified config.txt to use a custom device tree for Raspberry Pi 3 Model B. I placed the dtb file in Linux's /boot directory. It works. Then I tried to do the same for Debian arm64 buster. But it does not work. It still uses the builtin device tree. I tried to place the custom dtb file in the root of FAT partition. It does not either. What could be cause of this issue? Is there a way to get a detailed log for debugging.

Thanks.

pbatard, 2020-02-11:
Read the notes from https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/RPi3

__Please__ bear in mind that this repo is just for automated builds, therefore there's very little help I'm going to provide as everything derives from edk2-platform and you __will__ need to familiarize yourself with edk2-platforms if you want to do anything further than just use the published binaries.

closed 2020-02-11 by pbatard

jiez, 2020-02-11:
Thanks. I figured out the reason. When I installed Debian arm64 testing (bullseye), I created a custom.cfg for grub to tell grub the custom device tree. But I forgot that step when I for Debian arm64 buster. Actually I found those device_tree* lines in config.txt mentioned in the notes are not needed.