pftf/RPi3 #7: Wifi doesn't work at install time

e-minguez, 2020-02-20:
Thanks for your work here and the excellent blog post https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html
I've successfullly installed vanilla debian 10 following your instructions (using your latest release here) with ethernet but I cannot make wifi work at install time nor when it is installed. The `wlan0` device is there, wpa_supplicant configured, etc. but I cannot see any error log, it just doesn't connect.
I'm not sure if there is something missing wrt countries, channels, etc. Do you happen to have a successfull installation using wifi?
Thanks!

vemek, 2020-05-05:
Hey @e-minguez - after a very long evening of trying to figure this out, I tracked down the issue. The brcmfmac43430-sdio.txt shipped as part of these releases doesn't seem to be correct for the RPi 3B. You can fix this by copying [the file shipped by RPi-Distro](https://raw.githubusercontent.com/RPi-Distro/firmware-nonfree/master/brcm/brcmfmac43430-sdio.txt) to `/lib/firmware/brcm`. After a reboot, WiFi should work.

I'll cut a PR to update this.

vemek, 2020-05-05:
It looks like the existing config is using the same params as https://github.com/RPi-Distro/firmware-nonfree/blob/master/brcm/brcmfmac43430-sdio.AP6212.txt. I wonder if the chip changed with a board revision. Tomorrow I'll boot a fresh Rasbian install to see how they handle this, and cut a PR to attempt to do same.

pbatard, 2020-05-05:
So the __official__ armbian `brcmfmac43430-sdio.txt` from [here](https://github.com/armbian/firmware/tree/master/brcm) is improper?

There's a `brcmfmac43430a0-sdio.txt` there as well? Would that work better?

Ideally, I want to fetch the firmware files from well known official sources rather than obscure one, and I have no idea who's behind RPi-Distro. The commit where the file you sourced comes from contains no info about the file's origin (and contains changes for 870 files!), which isn't exactly ideal...

pbatard, 2020-05-05:
By the way, I'm probably going to update the binaries to the official 2020.04.02 version found [here](https://community.cypress.com/docs/DOC-19375) so it might be worth testing these files if you have a chance, thanks!

vemek, 2020-05-06:
> So the official armbian brcmfmac43430-sdio.txt from here is improper?
I'm beginning to suspect it's not improper (in that it works with other boards), but the parameters don't seem to work for the Pi 3B. I checked `brcmfmac43430a0-sdio.txt` and it won't work either - `boardrev` and `xtalfreq` are incorrect.

I agree about RPi-Distro; I stumbled upon them accidentally and likewise learned nothing from the commit history.

This evening I'll test against any more legit versions of `brcmfmac43430-sdio.txt` I can find, including the version shipped with stock Raspbian - I expect that will shed some light here, and at least tell us the "official" parameters we should be looking for from a trusted source.

I'll also test the new blobs from Cypress and confirm that they work :+1:

vemek, 2020-05-06:
I've verified that the file that works for me from that repo is the same one packaged by Raspbian - confirmed by extracting the [`firmware-brcm80211_20190114-1+rpt6_all.deb` from the Raspberry Pi apt repo](http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/).

I've checked all the firmware config files in the Armbian repo, and none target the same board revision that is on the Pi 3B - all their configs target `boardrev=0x1101` but we're dealing with `boardrev=0x1202`. Armbian doesn't target the Pi 3B, so this isn't all that surprising.

I'd suggest - if licencing permits - extracting the `brcmfmac43430-sdio.txt` from the Rasbian repo linked above as the source-of-truth here. I've tested this against the 2020-04-02 release from Cypress and it works perfectly with this config as well (and still fails for the existing config packaged in this repo).

I hope that makes sense. Let me know if there's anything else I can help test, and thanks a million for the original blog post / walkthrough that got me this far :)

pbatard, 2020-05-06:
> I'd suggest - if licencing permits - extracting the brcmfmac43430-sdio.txt from the Rasbian repo linked above as the source-of-truth here.

I think that's a good idea, because they're definitely the people best placed to know how what their hardware should use.

And licensing wise, these are proprietary files that comes with their separate license notice anyway, so it's not an issue. We're not embedding these files into any executables, so having a patchwork of licenses is no big deal, as long as we clearly state which is what for each file.

> I've tested this against the 2020-04-02 release from Cypress and it works perfectly with this config as well (and still fails for the existing config packaged in this repo).

Great! Thanks a lot for the testing. I'll go with what you suggest and try to update the files for the next release.

closed 2020-05-06 by pbatard

vemek, 2020-05-07:
One final note for posterity, which may be useful when Debian 11 is released: I've been experimenting with the Debian testing installer on my Pi 3B, and it looks like the broadcom driver in the testing repo explicitly names these config files by board, so it does _not_ pick up `brcmfmac43430-stdio.txt`. It instead is looking for the more explicit `brcm/brcmfmac43430-stdio.raspberrypi,3-model-b.txt`. I think this is a positive change as it reduces confusion around which config matches which board, but it's worth noting that you may want to ship that file under both names in future to maintain forwards- and backwards-compatibility.

e-minguez, 2020-05-11:
I can confirm the installation via wifi works now on my pi \o/ and after installation I just needed to configure my `/etc/network/interfaces` file to make wireless work.
Thanks!

pbatard, 2020-05-11:
@e-minguez, thanks for the report. And sorry it took us so long to address this!

e-minguez, 2020-05-11:
> @e-minguez, thanks for the report. And sorry it took us so long to address this!

No worries! it's been an interesting resolution :)

commit c8d17468 (2020-05-06): Update Wifi firmware to latest and fix 3B sdio.txt file

* Closes #7