pftf/RPi3 #14: Improve SDhost driver to fail gracefully when no card is present

andreiw, 2020-04-16:
Booting UEFI from USB or network still requires an SD card, because the SDhost driver will loop forever on I/O errors. Unfortunately the Pi Foundation didn't wire up the card detect pin so we have to do something more creative.

MagicRB, 2020-04-29:
A stupid solution would be try 200 or so times and then assume no card there

andreiw, 2020-04-30:
something like that, yeah

ehem, 2020-07-13:
The log in pftf/RPi4#77 seems interesting.  Notably that consistent "ERRI MmcStatus 0x18040" being an indicator of card absent.  The alternative which comes to mind is adding something to the command-line to tell it to skip trying MMC.

MagicRB, 2020-07-13:
I kind of maybe unrelated question, but from what I had gathered the in-CPU bootloader only looks for stuff to boot on the sdcard, so how can you get this UEFI bios running, without an sdcard?

pbatard, 2020-07-13:
> the in-CPU bootloader only looks for stuff to boot on the sdcard

That is not the case. At least for Bcm2711, which is the SoC used in the Pi 4, the on-die bootloader code is able to perform a lookup for boot files that reside on a USB drive. If you use a recent EEPROM with a recent `start4.elf`, you will get straight to USB boot.

MagicRB, 2020-07-13:
cool, i thought it didn't do that, you learn something new everyday ain't that right?

commit 06c36559 (2020-07-14): Update to EDK2 latest

* Closes #13
* Closes #14
closed 2020-07-14 by pbatard