pftf/RPi3 #18: Unable to boot from USB storage

kailiu42, 2020-05-12:
Previously I boot from SD card and it's OK, except it's slow. So I decided to boot from a fast USB drive I have.

After I have the USB drive set up, and have the device USB boot mode enabled per [this document](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md), the bootloader successfully booted (as I could see the multi-colored screen). But the UEFI boot logo (the black and white raspberry) never showed. Here is what's on the serial output:

```
Raspberry Pi Bootcode
Read File: config.txt, 399
Read File: start.elf, 2976448 (bytes)
Read File: fixup.dat, 7270 (bytes)
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.3():v2.3
NOTICE:  BL1: Built : 10:40:46, Apr 21 2020
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Embest, China) [0x00a22082]
NOTICE:  BL1: Booting BL2
ERROR:   rpi3_sdhost: timeout status 0x40
NOTICE:  BL2: v2.3():v2.3
NOTICE:  BL2: Built : 10:40:42, Apr 21 2020
NOTICE:  BL1: Booting BL31
NOTICE:  BL31: v2.3():v2.3
NOTICE:  BL31: Built : 10:40:44, Apr 21 2020
UEFI firmware (version UEFI Firmware v1.24 built at 16:56:19 on May  6 2020)
```

And there was not more output, no more activity on the HDMI screen. No any further changes if I press keys on keyboard or on serial console.

If I have the bootable SD card plugged in with the USB drive together, I can use the UEFI firmware booted from the SD to load GRUB from USB drive then boot my OS.

pbatard, 2020-05-12:
At the moment, the UEFI firmware requires an SD card to be present, because the Arasan driver loops forever on I/O error (which it gets if no card is present).

This is being tracked in pftf/RPi4#37 (the driver is common to RPi3 and RPi4, so the RPi4 fix will also apply to RPi3).

Considering that this is a duplicate of something we are actively tracking, I will close this issue.

pbatard, 2020-05-12:
Actually, we also have #13 and #14, that are opened in this repository, that are also tracking the same.

closed 2020-05-12 by pbatard