pftf/RPi3 #21: Could pftf/RPi3 have Serial/Graphical selection like pftf/RPi4? (Example use: FreeBSD context)

markmi, 2020-09-14:
Note: I start out with a comparison to FreeBSD use with pftf/RPi4. But this is not a request for pftf/RPi4 changes.

FreeBSD crashes in early kernel activity for Graphical for pftf/RPi4 but is useful with Serial selected (log in, build FreeBSD, etc.). (I use disable-bt .)

Turns out FreeBSD crashes in the same way for pftf/RPi3 (on a RPi3B v1.2) as for Graphical with pftf/RPi4 --but there is no explicit/alternative selection present in pftf/RPi3: implicitly/always Graphical, or so it appears. I expect that, with Serial available to select and save, the result would work like it does for pftf/RPi4 .

Beyond the above:
The crash is such that USB input to the kernel prompt is non-functional and the serial console is not available (no input, output stopped earlier in the sequence when it switched to Graphical). This makes capturing the messages leading to the kernel-backtrace difficult to capture. (I do not claim to know how this fits with pftf/RPI3 vs. the FreeBSD kernel vs. both but it is a difficulty for evidence gathering.) With a sustained serial output stream I'd be able to just record it and then report it on a FreeBSD list.

markmi, 2020-09-14:
In case it is of interest, an example of the last text to go out on the serial console follows. There is more output generated by the kernel afterwards but it does not appear on the serial console. (The notes about "device tree blob" not being found are normal for FreeBSD with ACPI selected.)

```
Hit [Enter] to boot immediately, or any other key for command prompt.
Booting [/boot/kernel/kernel]...               
No valid device tree blob found!
WARNING! Trying to fire up the kernel, but no device tree blob found!
EFI framebuffer information:
addr, size     0x3e402000, 0x7e9000
dimensions     1920 x 1080
stride         1920
masks          0x00ff0000, 0x0000ff00, 0x000000ff, 0xff000000
```

samerhaj, 2020-09-17:
Should be possible by just adding ConsolePrefDxe to the RPi3 DSC/FDF.. Will submit a patch

bzfbd, 2021-10-03:
set console="comconsole" instead of console=efi does get you a bit further (the HDMI will continue a bit longer) but the serial console will start scrambling somewhere have the way through the kernel boot.