pftf/RPi3 #22: Boot stalls after upgrade

matthias-t, 2020-09-29:
I recently upgraded the UEFI firmware on my Pi 3 B+, by clearing my microSD and unpacking your release archive onto it. (Until now, I was using a version from before it was merged into the tianocore repositories.) When booting my current NixOS system, I get:

```
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Loaded initrd from command line option
EFI stub: Exiting boot services and installing virtual address map...
EFI stub: ERROR: Unable to construct new device tree.
EFI stub: ERROR: Failed to update FDT and exit boot services
```

When trying to boot a debian live stick, the firmware interface shows up, and then GRUB, but after selecting `* Install` the boot process stalls and all I get is a cursor in the top left of the screen.

I'm guessing this has something to do with device trees? In the debian stick's GRUB command-line, typing `devicetree` yields nothing.

pbatard, 2020-10-05:
Confirmed the issue with the Debian installer.
There appears to be a regression between 1.29 and 1.30 that prevents Linux from booting on the Pi 3. Will have to pinpoint the commit before we can address it...

pbatard, 2020-10-06:
Oh, great, it's not the UEFI firmware, but it's the Raspberry Pi Foundation having broken `start.elf`/`fixup.dat` for the Pi 3 yet again...

If you use the 1.30 UEFI firmware with the 1.29 support files, everything is fine. But as soon as you try to use more recent `start.elf`/`fixup.dat`  all hell breaks loose. For one thing (with the Debian installer kernel), if using the version of the support files embedded in the 1.30 archive, the whole boot process gets stuck. And if you use the very latest version at the time of this post, you get GRUB error `error: failed to install/update FDT.` before it sends you back to the GRUB prompt...

pbatard, 2020-10-06:
Testing shows that this is a continuation of raspberrypi/firmware#1445, as using the July 31st version of the Pi firmware files is fine, but the August 5th version has the new regression. Will have to see what the Pi Foundation people have to say about it. In the meantime, I have updated the zip archive to use the July 31st version of the Pi firmware files.

pbatard, 2020-10-06:
And now that I have the new baudrate, I can report that the new version of `start.elf` results in [an assert in `FtdDxe.c`](https://github.com/tianocore/edk2-platforms/blob/d492639638eee331ac3389e6cf53ea266c3c84b3/Platform/RaspberryPi/Drivers/FdtDxe/FdtDxe.c#L256). So it would look like `fdt_add_subnode()` fails to add a `psci` node?

pbatard, 2020-10-12:
haven't had much chance to look further into this yet, but here's what I have so far:

* Works (old `start.elf`):
```
Devicetree passed via config.txt (0x73EB bytes)
/psci lookup: -1
/psci addon: 208
```
* Fails (new `start.elf`):
```
Devicetree passed via config.txt (0x6FB7 bytes)
/psci lookup: -11
/psci addon: -11
ASSERT [FdtDxe] /usr/src/edk2-platforms/Platform/RaspberryPi/Drivers/FdtDxe/FdtDxe.c(258): Node >= 0
```

Of course, the `.dtb` was exactly the same for both these tests, and its size **is** 28599 bytes, i.e. 0x6FB7 bytes, so with the old working configuration, the Device Tree size we retrieved was actually incorrect?!?

pbatard, 2020-10-13:
So, `-11` is `FDT_ERR_BADSTRUCTURE`, which means that, according to `edk2\EmbeddedPkg\Library\FdtLib`, the _"device tree has a corrupt structure block or other serious error (e.g. misnested nodes, or subnodes preceding properties)"._

I am now looking at the DT dumps for both instances, and it looks like the DT we get with the newer `start.elf`, while seemingly closer to the original binary DT, is altered in some breaking fashion:
```
Devicetree passed via config.txt (0x6FB7 bytes)
  00000000  D0 0D FE ED 00 00 8F B7 00 00 00 48 00 00 64 84  ...........H..d.
  00000010  00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00  ...(............
  00000020  00 00 0B 33 00 00 64 3C 00 00 00 00 00 00 00 00  ...3..d<........
  00000030  00 00 00 00 00 00 10 00 00 00 00 00 00 00 00 00  ................
  00000040  00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00  ................
  00000050  00 00 00 03 00 00 00 08 00 00 0B A5 08 00 00 00  ................
  00000060  62 65 72 72 79 70 69 2C 33 2D 6D 6F 64 65 6C 2D  berrypi,3-model-
  00000070  62 2D 70 6C 75 73 00 62 72 63 6D 2C 62 63 6D 32  b-plus.brcm,bcm2
  00000080  38 33 37 00 00 00 00 03 00 00 00 18 00 00 00 0B  837.............
  00000090  52 61 73 70 62 65 72 72 79 20 50 69 20 33 20 4D  Raspberry Pi 3 M
  000000A0  6F 64 65 6C 20 42 2B 00 00 00 00 03 00 00 00 04  odel B+.........
  000000B0  00 00 00 11 00 00 00 01 00 00 00 03 00 00 00 04  ................
  000000C0  00 00 00 20 00 00 00 01 00 00 00 03 00 00 00 04  ... ............
  000000D0  00 00 00 2C 00 00 00 01 00 00 00 01 61 6C 69 61  ...,........alia
  000000E0  73 65 73 00 00 00 00 03 00 00 00 12 00 00 0B 4E  ses............N
  000000F0  2F 73 6F 63 2F 69 32 63 40 37 65 38 30 34 30 30  /soc/i2c@7e80400
  00000100  35 30 34 30 00 00 00 00 00 00 00 03 00 00 00 15  5040............
  00000110  00 00 00 45 2F 73 6F 63 2F 73 65 72 69 61 6C 40  ...E/soc/serial@
  00000120  37 65 32 30 31 30 30 30 00 00 00 00 00 00 00 03  7e201000........
  00000130  00 00 00 24 00 00 00 4D 2F 73 6F 63 2F 6D 61 69  ...$...M/soc/mai
  00000140  6C 62 6F 78 40 37 65 30 30 62 38 34 30 2F 62 63  lbox@7e00b840/bc
  00000150  6D 32 38 33 35 5F 61 75 64 69 6F 00 00 00 00 03  m2835_audio.....
  00000160  35 30 34 30 00 00 00 00 00 00 00 03 00 00 00 15  5040............
  00000170  00 00 00 45 2F 73 6F 63 2F 73 65 72 69 61 6C 40  ...E/soc/serial@
  00000180  00 00 00 0B 00 00 00 57 2F 73 6F 63 2F 73 6F 75  .......W/soc/sou
  00000190  6E 64 00 00 00 00 00 03 00 00 00 05 00 00 00 5D  nd.............]
  000001A0  2F 73 6F 63 00 00 00 00 00 00 00 03 00 00 00 12  /soc............
  000001B0  00 00 00 61 2F 73 6F 63 2F 64 6D 61 40 37 65 30  ...a/soc/dma@7e0
  000001C0  30 37 30 30 30 00 00 00 00 00 00 03 00 00 00 23  07000..........#
  000001D0  00 00 00 65 2F 73 6F 63 2F 69 6E 74 65 72 72 75  ...e/soc/interru
  000001E0  70 74 2D 63 6F 6E 74 72 6F 6C 6C 65 72 40 37 65  pt-controller@7e
  000001F0  30 30 62 32 30 30 00 00 00 00 00 03 00 00 00 17  00b200..........
  00000200  00 00 00 6A 2F 73 6F 63 2F 77 61 74 63 68 64 6F  ...j/soc/watchdo
  00000210  67 40 37 65 31 30 30 30 30 30 00 00 00 00 00 03  g@7e100000......
  00000220  00 00 00 12 00 00 00 73 2F 73 6F 63 2F 72 6E 67  .......s/soc/rng
  00000230  40 37 65 31 30 34 30 30 30 00 00 00 00 00 00 03  @7e104000.......
  00000240  00 00 00 16 00 00 00 7A 2F 73 6F 63 2F 6D 61 69  .......z/soc/mai
  00000250  6C 62 6F 78 40 37 65 30 30 62 38 38 30 00 00 00  lbox@7e00b880...
  00000260  00 00 00 03 00 00 00 13 00 00 00 82 2F 73 6F 63  ............/soc
  00000270  2F 67 70 69 6F 40 37 65 32 30 30 30 30 30 00 00  /gpio@7e200000..
  00000280  00 00 00 03 00 00 00 15 00 00 00 87 2F 73 6F 63  ............/soc
  00000290  2F 73 65 72 69 61 6C 40 37 65 32 30 31 30 30 30  /serial@7e201000
  000002A0  00 00 00 00 00 00 00 03 00 00 00 15 00 00 00 8D  ................
  000002B0  2F 73 6F 63 2F 73 65 72 69 61 6C 40 37 65 32 31  /soc/serial@7e21
  000002C0  35 30 34 30 00 00 00 00 00 00 00 03 00 00 00 12  5040............
  000002D0  00 00 00 93 2F 73 6F 63 2F 6D 6D 63 40 37 65 32  ..../soc/mmc@7e2
  000002E0  00 00 00 03 00 00 00 15 00 00 00 87 2F 73 6F 63  ............/soc
  000002F0  2F 73 65 72 69 61 6C 40 37 65 32 30 31 30 30 30  /serial@7e201000
  00000300  00 00 00 03 00 00 00 17 00 00 00 6A 2F 73 6F 63  ...........j/soc
  00000310  2F 77 61 74 63 68 64 6F 67 40 37 65 31 30 30 30  /watchdog@7e1000
  00000320  30 30 00 00 00 00 00 03 00 00 00 12 00 00 00 73  00.............s
  00000330  2F 73 6F 63 2F 72 6E 67 40 37 65 31 30 34 30 30  /soc/rng@7e10400
  00000340  30 32 30 30 30 00 00 00 00 00 00 03 00 00 00 12  02000...........
  00000350  00 00 00 9A 2F 73 6F 63 2F 6D 6D 63 40 37 65 33  ..../soc/mmc@7e3
  00000360  30 30 30 30 30 00 00 00 00 00 00 03 00 00 00 14  00000...........
  00000370  00 00 00 9E 2F 73 6F 63 2F 6D 6D 63 6E 72 40 37  ..../soc/mmcnr@7
  00000380  65 33 30 30 30 30 30 00 00 00 00 03 00 00 00 12  e300000.........
  00000390  00 00 00 A3 2F 73 6F 63 2F 6D 6D 63 40 37 65 32  ..../soc/mmc@7e2
  000003A0  30 34 30 30 30 00 00 00 00 00 00 03 00 00 00 13  04000...........
  000003B0  00 00 00 B6 2F 73 6F 63 2F 69 32 63 30 6D 75 78  ..../soc/i2c0mux
  000003C0  2F 69 32 63 40 31 00 00 00 00 00 03 00 00 00 12  /i2c@1..........
  000003D0  00 00 00 BC 2F 73 6F 63 2F 73 70 69 40 37 65 32  ..../soc/spi@7e2
  000003E0  2F 6D 6D 63 40 37 65 32 30 32 30 30 30 00 00 00  /mmc@7e202000...
  000003F0  00 00 00 03 00 00 00 12 00 00 00 9A 2F 73 6F 63  ............/soc
[... cut ...]
  000007C0  72 76 65 64 2D 6D 65 6D 6F 72 79 00 00 00 00 03  rved-memory.....
  000007D0  00 00 00 04 00 00 00 11 00 00 00 01 00 00 00 03  ................
  000007E0  00 00 00 04 00 00 00 20 00 00 00 01 00 00 00 03  ....... ........
  000007F0  00 00 00 00 00 00 00 FF 00 00 00 03 00 00 00 04  ................
```

Especially, that truncated `reserved-memory` near the end doesn't seem very auspicious and is unlikely to be valid. So I would wager that the newer `start.elf` have tasked VideoCore to alter the DT in a manner that is breaking it...

pbatard, 2020-10-15:
Confirmed it's a `start.elf` bug, as the new test version from raspberrypi/firmware#1445 appears to fix the issue (DT reported size has been updated and no more truncated `reserved-memory` is seen) and the Debian installer launches as expected:
```
Devicetree passed via config.txt (0x73EB bytes)
  00000000  D0 0D FE ED 00 00 93 EB 00 00 00 48 00 00 68 14  ...........H..h.
  00000010  00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00  ...(............
  00000020  00 00 0B D7 00 00 67 CC 00 00 00 00 00 00 00 00  ......g.........
  00000030  00 00 00 00 00 00 10 00 00 00 00 00 00 00 00 00  ................
  00000040  00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00  ................
  00000050  00 00 00 03 00 00 00 08 00 00 0B A5 3B 40 00 00  ............;@..
  00000060  04 C0 00 00 00 00 00 03 00 00 00 11 00 00 0B 97  ................
  00000070  30 30 30 30 30 30 30 30 62 62 38 33 35 37 33 64  00000000bb83573d
  00000080  00 65 6C 2D 00 00 00 03 00 00 00 28 00 00 00 00  .el-.......(....
  00000090  72 61 73 70 62 65 72 72 79 70 69 2C 33 2D 6D 6F  raspberrypi,3-mo
  000000A0  64 65 6C 2D 62 2D 70 6C 75 73 00 62 72 63 6D 2C  del-b-plus.brcm,
  000000B0  62 63 6D 32 38 33 37 00 00 00 00 03 00 00 00 24  bcm2837........$
  000000C0  00 00 00 0B 52 61 73 70 62 65 72 72 79 20 50 69  ....Raspberry Pi
  000000D0  20 33 20 4D 6F 64 65 6C 20 42 20 50 6C 75 73 20   3 Model B Plus
  000000E0  52 65 76 20 31 2E 33 00 00 00 00 03 00 00 00 04  Rev 1.3.........
  000000F0  00 00 00 11 00 00 00 01 00 00 00 03 00 00 00 04  ................
  00000100  00 00 00 20 00 00 00 01 00 00 00 03 00 00 00 04  ... ............
  00000110  00 00 00 2C 00 00 00 01 00 00 00 01 73 79 73 74  ...,........syst
  00000120  65 6D 00 00 00 00 00 03 00 00 00 08 00 00 0B BF  em..............
  00000130  00 00 00 00 BB 83 57 3D 00 00 00 03 00 00 00 04  ......W=........
  00000140  00 00 0B B0 00 A0 20 D3 00 00 00 02 00 00 00 01  ...... .........
  00000150  61 78 69 00 00 00 00 01 76 63 5F 6D 65 6D 00 00  axi.....vc_mem..
  00000160  00 00 00 03 00 00 00 0C 00 00 01 7A 3E C0 00 00  ...........z>...
  00000170  40 00 00 00 C0 00 00 00 00 00 00 02 00 00 00 02  @...............
  00000180  00 00 00 01 61 6C 69 61 73 65 73 00 00 00 00 03  ....aliases.....
  00000190  00 00 00 12 00 00 0B 4E 2F 73 6F 63 2F 69 32 63  .......N/soc/i2c
  000001A0  40 37 65 38 30 34 30 30 30 00 00 30 00 00 00 03  @7e804000..0....
  000001B0  00 00 00 12 00 00 0B 4A 2F 73 6F 63 2F 69 32 63  .......J/soc/i2c
  000001C0  40 37 65 38 30 34 30 30 30 00 00 30 00 00 00 03  @7e804000..0....
  000001D0  00 00 00 13 00 00 0B 33 2F 73 6F 63 2F 69 32 63  .......3/soc/i2c
  000001E0  30 6D 75 78 2F 69 32 63 40 30 00 30 00 00 00 03  0mux/i2c@0.0....
  000001F0  00 00 00 15 00 00 00 3D 2F 73 6F 63 2F 73 65 72  .......=/soc/ser
  00000200  69 61 6C 40 37 65 32 31 35 30 34 30 00 00 00 00  ial@7e215040....
  00000210  00 00 00 03 00 00 00 15 00 00 00 45 2F 73 6F 63  ...........E/soc
  00000220  2F 73 65 72 69 61 6C 40 37 65 32 30 31 30 30 30  /serial@7e201000
  00000230  00 00 00 00 00 00 00 03 00 00 00 24 00 00 00 4D  ...........$...M
  00000240  2F 73 6F 63 2F 6D 61 69 6C 62 6F 78 40 37 65 30  /soc/mailbox@7e0
  00000250  30 62 38 34 30 2F 62 63 6D 32 38 33 35 5F 61 75  0b840/bcm2835_au
  00000260  64 69 6F 00 00 00 00 03 00 00 00 12 00 00 00 53  dio............S
  00000270  2F 73 6F 63 2F 61 75 78 40 37 65 32 31 35 30 30  /soc/aux@7e21500
  00000280  30 00 00 00 00 00 00 03 00 00 00 0B 00 00 00 57  0..............W
  00000290  2F 73 6F 63 2F 73 6F 75 6E 64 00 00 00 00 00 03  /soc/sound......
  000002A0  00 00 00 05 00 00 00 5D 2F 73 6F 63 00 00 00 00  .......]/soc....
  000002B0  00 00 00 03 00 00 00 12 00 00 00 61 2F 73 6F 63  ...........a/soc
  000002C0  2F 64 6D 61 40 37 65 30 30 37 30 30 30 00 00 00  /dma@7e007000...
  000002D0  00 00 00 03 00 00 00 23 00 00 00 65 2F 73 6F 63  .......#...e/soc
  000002E0  2F 69 6E 74 65 72 72 75 70 74 2D 63 6F 6E 74 72  /interrupt-contr
  000002F0  6F 6C 6C 65 72 40 37 65 30 30 62 32 30 30 00 00  oller@7e00b200..
  00000300  00 00 00 03 00 00 00 17 00 00 00 6A 2F 73 6F 63  ...........j/soc
  00000310  2F 77 61 74 63 68 64 6F 67 40 37 65 31 30 30 30  /watchdog@7e1000
  00000320  30 30 00 00 00 00 00 03 00 00 00 12 00 00 00 73  00.............s
  00000330  2F 73 6F 63 2F 72 6E 67 40 37 65 31 30 34 30 30  /soc/rng@7e10400
  00000340  30 00 00 00 00 00 00 03 00 00 00 16 00 00 00 7A  0..............z
  00000350  2F 73 6F 63 2F 6D 61 69 6C 62 6F 78 40 37 65 30  /soc/mailbox@7e0
  00000360  30 62 38 38 30 00 00 00 00 00 00 03 00 00 00 13  0b880...........
  00000370  00 00 00 82 2F 73 6F 63 2F 67 70 69 6F 40 37 65  ..../soc/gpio@7e
  00000380  32 30 30 30 30 30 00 00 00 00 00 03 00 00 00 15  200000..........
  00000390  00 00 00 87 2F 73 6F 63 2F 73 65 72 69 61 6C 40  ..../soc/serial@
  000003A0  37 65 32 30 31 30 30 30 00 00 00 00 00 00 00 03  7e201000........
  000003B0  00 00 00 15 00 00 00 8D 2F 73 6F 63 2F 73 65 72  ......../soc/ser
  000003C0  69 61 6C 40 37 65 32 31 35 30 34 30 00 00 00 00  ial@7e215040....
  000003D0  00 00 00 03 00 00 00 12 00 00 00 93 2F 73 6F 63  ............/soc
  000003E0  2F 6D 6D 63 40 37 65 32 30 32 30 30 30 00 00 00  /mmc@7e202000...
  000003F0  00 00 00 03 00 00 00 12 00 00 00 9A 2F 73 6F 63  ............/soc
  00000400  2F 6D 6D 63 40 37 65 33 30 30 30 30 30 00 00 00  /mmc@7e300000...
  00000410  00 00 00 03 00 00 00 14 00 00 00 9E 2F 73 6F 63  ............/soc
[... cut ...]
  00000790  6D 6D 63 62 6C 6B 30 70 32 20 72 6F 6F 74 66 73  mmcblk0p2 rootfs
  000007A0  74 79 70 65 3D 65 78 74 34 20 72 6F 6F 74 77 61  type=ext4 rootwa
  000007B0  69 74 00 72 00 00 00 02 00 00 00 01 72 65 73 65  it.r........rese
  000007C0  72 76 65 64 2D 6D 65 6D 6F 72 79 00 00 00 00 03  rved-memory.....
  000007D0  00 00 00 04 00 00 00 11 00 00 00 01 00 00 00 03  ................
  000007E0  00 00 00 04 00 00 00 20 00 00 00 01 00 00 00 03  ....... ........
  000007F0  00 00 00 00 00 00 00 FF 00 00 00 03 00 00 00 04  ................
```

pbatard, 2020-10-15:
This is now fixed with the latest update from https://github.com/raspberrypi/firmware/tree/master/boot. We'll pick it up with the next firmware release.

closed 2020-10-15 by pbatard

matthias-t, 2020-10-15:
Thanks a lot for identifying the issue so quickly and reporting it upstream! I'm really impressed by your work.

On an unrelated note, I wrote a Nix expression to build the EDK II firmware, so if you're interested in using Nix in your CI instead of shell, I'm happy to submit a PR. Alternatively, I might try to get it merged into Nixpkgs, in which case the [Hydra](https://hydra.nixos.org/) will build it too.

robgiorgi, 2023-02-24:
....I experience the same problem when switching from kernel 6.0 to 6.1 (Fedora 37). What is the solution to make 6.1 working?