pftf/RPi3 #33: Rainbow screen and  console print error

piaoxuebingfeng, 2021-06-16:
Raspberry Pi Bootcode

Read File: config.txt, 161

Read File: start.elf, 2955648 (bytes)

Read File: fixup.dat, 7311 (bytes)

MESS:00:00:00.968554:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:00.972876:0: brfs: File read: 161 bytes
MESS:00:00:01.056792:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.061068:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.884477:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:01.892006:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.897941:0: *** Restart logging
MESS:00:00:01.901819:0: brfs: File read: 161 bytes
MESS:00:00:01.939160:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:01.948512:0: dtb_file 'bcm2710-rpi-3-b.dtb'
MESS:00:00:01.956749:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
MESS:00:00:01.961404:0: Loading 'bcm2710-rpi-3-b.dtb' to 0x1f0000 size 0x6ee8
MESS:00:00:01.981760:0: brfs: File read: 28392 bytes
MESS:00:00:02.051013:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.061043:0: brfs: File read: 161 bytes
MESS:00:00:02.064278:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:02.086425:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:03.999652:0: brfs: File read: /mfs/sd/RPI_EFI.fd
MESS:00:00:04.003509:0: Loading 'RPI_EFI.fd' to 0x0 size 0x1f0000
MESS:00:00:04.009333:0: No compatible kernel found
MESS:00:00:04.013835:0: Device tree loaded to 0x1f0000 (size 0x7320)
MESS:00:00:04.020943:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:04.027601:0: uart: Baud rate change done...
MESS:00:00:04.031015:0: uart: Baud rateNOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.3():v2.3
NOTICE:  BL1: Built : 10:40:46, Apr 21 2020
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
NOTICE:  BL1: Booting BL2
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd
ERROR:   rpi3_sdhost status: 0x10
ERROR:   rpi3_sdhost: unknown err, cmd = 0xd

how to reslove this error?

pbatard, 2021-07-04:
Have you tried with a different SD card? This looks like a defective SD card issue.

pbatard, 2021-07-04:
Also, this seems a duplicate of #17.

closed 2021-07-04 by pbatard