pftf/RPi3 #35: Debian 11 on RPi3B

high1, 2021-09-20:
After following the https://www.raspberrypi.org/forums/viewtopic.php?f=50&t=249449&sid=603ea55543a7804affa4f421ecd32c64 guide,  installation goes as planned but reboot fails with Synchronous Exception at 0x17FFFFFA9402E831. Is there something that I've missed that should be done for Debian 11 differently? It's a plain install, without desktop environment. EFI starts up, blinks the GRUB prompt and then the exception happens. Would appreciate any kind of help.

qyz, 2021-09-29:
I can confirm this exact problem. Installation using the Debian 11 netinst-ISO runs smoothly but the installed system fails to boot.

pgzh, 2021-10-06:
I am seeing the same error and have a feeling it may be grub-related since it happens when grub should be showing.
Downgrading to the previous release of the uefi firmware after the installation (by overwriting all the files in the EFI partition) didn't change anything.

Unrelated issue I encountered:
Also I've had some trouble installing via netinst, I couldn't manage to get a wired network connection - I've tried different ports on my switch, changed cables and also tried with a different Pi.
In the end I managed to get a connection by doing a modprobe -r lan78xx and loading the module again from the installer console.
Maybe this should be mentioned in the documentation in case it's an issue with Debian 11.

schattenpinguin, 2021-11-02:
I´m facing the same issue on my Raspi 3b with Debian 11.
Installation itself ran without unexpected issues, just what was listed in documenteation linkes by high1 already.

I really would prefer installing Debian 11 on my own, as "tested Debian Images" show slightly differences in configuration and installed packages. This is annoying as i try to setup identical systems on x86_64 VM and ARM_64 Raspberry.

josteink, 2022-02-15:
Not sure if that's relevant, but I'm currently running plain Debian 11 on my Rpi 3B+ using this firmware.

Note: It was installed as Debian 10, and has sub-sequently been dist-upgraded.

farfade, 2022-02-15:
Hello,

Same here :(

Installation goes as planned but reboot fails with Synchronous Exception at 0x17FFFFFA9402E831 at the moment the grub loader should take the lead.

paulwratt, 2022-02-25:
IIRC Action Retro had the same "installs fine but fails to boot" issue with Debian 11 on PPC. _That may_ be for a different reason, but it is suspiciously similar.

lurkerpro, 2022-06-25:
I used "Device Tree" or "DT mode" and it didn't work. I downgraded to Debian 10 ("buster")  and it could successfully boot after installation. So I guess it's not related to it. 

Is it possible some firmware didn't load successfully during installation that caused "Synchronous Exception at 0x17FFFFFA9402E831"?

I followed the https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html for most steps
However, for 

> Though they are included in the UEFI firmware zip archive we linked above, it is most likely okay not to provide the .clm_blob if you don't have it (the Wifi drivers should work without that file), so don't be afraid to select No here if needed.

One thing different I noticed for "[bullseye](https://www.debian.org/releases/bullseye/)"and "[buster](https://www.debian.org/releases/buster/)" is that, when I was installing buster, all firmware installed successfully, while bullseye always prompting for one file it didn't find. 

Is it possible that this firmware somehow related to network interface which caused grub loader to fail?

Is anyone here able to confirm my guess by changing the filename to what bullseye is asking for and check if it could boot after installation?

lurkerpro, 2022-06-25:
> Hello,
> 
> Same here :(
> 
> Installation goes as planned but reboot fails with Synchronous Exception at 0x17FFFFFA9402E831 at the moment the grub loader should take the lead.

I am not sure why grub loader Exception so I deleted the config grub use so it fallback to some kind of command line prompt. And it didn't crash. 

After I entered the prompt I could source the same config and reproduce the issue. `Synchronous Exception at 0x17FFFFFA9402E831`

I suspected it might be a config issue, so I added some log to the file and boot multiple time, and it not always crash at the same line. 
So I guess it might somehow related to hardware? Like the firmware issue I suggested above?
I am not very familiar with hardware, so please correct me if I am wrong.

vemek, 2022-07-19:
I spent some time figuring this out and have discovered that the issue is in the version of `shim-signed` that ships with Debian 11. To correct it, enter a chroot of your installation (either from install media or by mounting the SD card on another host) and run
```
apt purge shim-signed
```
This should trigger grub to reinstall as per your previous configuration. You can confirm this is the case by looking in `/boot/efi/EFI/debian` or `/boot/efi/EFI/boot` - there should be no `shimaa64.efi`. If there is, remove that directory and run
```
grub-install --target=arm64-efi --efi-directory=/boot/efi     --bootloader-id=debian --recheck --no-floppy --removable
```

This fixes the Synchronous Exception error.

After fixing this, I still had issues with grub hanging after install the virtual address map. This seems to have been fixed by changing from ACPI mode to ACPI + DT.

schattenpinguin, 2022-08-25:
Hi vemek,
i tried your steps, but it´s not working for me.

I used a different SD card on same raspberry, with debian "native" image.
Vanilla image with "Synchronous Exception" error was connected via USB-SD-adapter:
/dev/sda1 - EFI
/dev/sda2 - swap
/dev/sda3 - /

I prepared chroot like:
mount /dev/sda3 /mnt
mount /dev/sda1 /mnt/boot/efi
mount --rbind /dev  /mnt/dev
mount --rbind /proc /mnt/proc
mount --rbind /sys  /mnt/sys
chroot /mnt

Then i renamed both folder
mv /boot/efi/EFI/debian /boot/efi/EFI/debian.org
mv /boot/efi/EFI/boot /boot/efi/EFI/boot.org

Only then grub-install created new folder
/boot/efi/EFI/BOOT
yes, capital letter. Folder "debian" is not created at all.

When i reboot the system, grub shows the menu to install/graphical install/...
So it boots into Netinstall-ISO on EFI partition, instead of my root.

vemek, 2022-08-25:
> When i reboot the system, grub shows the menu to install/graphical install/...
> So it boots into Netinstall-ISO on EFI partition, instead of my root.

Sorry @schattenpinguin, I don't understand your setup. Where is the netinstall image coming from? Look for files called `grub.cfg` in `/boot` - there may be some leftover from the install if you used the same media for install and target. Running `grub-update` in the chroot may fix this.

schattenpinguin, 2022-08-26:
I also followed
https://pete.akeo.ie/2019/07/installing-debian-arm64-on-raspberry-pi.html
to prepare SD card "1". So first partition (EFI) includes uefi-firmware and content from debian netinstall-iso.
During debian installation i left EFI partition untouched.

Then i used SD card "2" to mount above card for chroot.
Card 2 has been setup according to
https://raspi.debian.net/
Of course this is placed into SD-Slot of my Raspberry.
So i used USB-SD-Adapter to connect Card "1", like this disk is recognized as /dev/sda.