pftf/RPi3 #41: Support for Zero 2?

1nikolas, 2022-12-22:
As far as I know, the Raspberry PI Zero 2 has the same SoC as the 3 lineup. I tried to use the current version of the firmware on my Ζero 2 W. It boots to the setup/efi shell but can't read the SD card. Is this something that can be easily fixed?
Thanks

The SoC is BCM2710A1 as far as I know
[image]

Selecting this does nothing
[image]

Boot from file finds no partition
[image]

alsldlflglhljlk, 2024-04-22:
Does your SD card have an ESP formatted as FAT32?

1nikolas, 2024-04-24:
> Does your SD card have an ESP properly formatted as FAT32?

I don't even remember what I did one year ago 😅 I'll check again some time

keixthb, 2025-08-27:
@bradfa @1nikolas How were you able to achieve this? I've tried formatting a micro sd card with both fat32 and fat16 partitions, as well as a USBA drive, and was unable to load the EFI screen. I tried the latest version (v1.39), a previous version (v1.38) and an older version (v1.29). Each version tested will consistently boot to the rainbow screen, but does not proceed to the EFI.

I'm using the msdos (mbr) partition table.  

Surely, I must be doing something wrong. Can you please help me? I'm also trying to use the Raspberry pi zero 2w.

closed 2026-08-31 by ardbiesheuvel