pftf/RPi3 #43: Unable to boot FreeBSD EFI

DAZ-NZ-Eng, 2023-08-06:
I have a FreeBSD image on an USB stick. I can setup a VM with UEFI mode enabled and attached the USB as a SATA physical disk to the VM and it boots up just fine. However, With pftf/RPi3B+ burnt to an SD card and attaching the USB with FreeBSD to the raspberry pi 3B+, I am unable to boot FreeBSD. 
Is there a recommend settings to enable to boot?

bzfbd, 2023-08-12:
For UEFI boot I think you just create a GPT with gpart, add an -t efi for partition 1;  there goes the firmware and as you say EFI/BOOT/bootaa64.efi.  Then add a -t freebsd-ufs partition and do the normal install there.   For SD this definitively works.  Depending on FreeBSD release and whether you use HDMI or serial you may have to set different options.

DAZ-NZ-Eng, 2023-08-13:
I followed Bjorn's guide.
I am using pftf V1.39, FreeBSD-13.2-RELEASE-arm64-aarch64-disc1.iso.

When FreeBSD OS initiates, I noted that:
 ue0: <usb ethernet> on muge0 raspberry pi
.
.
.
ue0: link state changed to down

When it reached this point of the boot, It consistently freezes at this line every time.

However, I have managed to boot as root when I select SAFE mode and SINGLE USER in FreeBSD initial boot option.