pftf/RPi3 #46: Fedora CoreOS: stuck on efi stub: exiting boot services

Aegeontis, 2024-03-28:
1. Installed UEFI on an sdcard
2. Flashed a fedora coreos (arm64) image to a usb drive
3. As per the readme, manually edited the cmdline in grub and added `acpi=off` (for some reason the uefi settings refused to save that option :shrug:).
4. Press f10 to boot
5. I get exactly the following output:
```
Booting a command list

EFI stub: Decompressing Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...
```
6. The usb continues flashing (i.e. either something is writing to it or something is reading from it), but nothing further happens. I let it sit for at least 10 minutes, but still nothing.

alsldlflglhljlk, 2024-04-20:
Try pressing `F1` to enter the UEFI shell, then type `fs0:`, `fs1:`, `fs2:` etc until you find where your kernel is located. Then type:
`.\kernelnamehere root=/dev/rootdevicehere rootwait ro` and see what happens.

Aegeontis, 2024-04-21:
For some reason I couldnt find the kernel and rootfs directly in any of the locations listed by ``map`` in the UEFI shell (only an EFI folder on fs0 with grub binaries). Therefore I did the following: 
1. created a new usb drive from my desktop and manually copied just the rootfs.img and vmlinuz file to it. 
2. Then I plugged it into the rpi, cd'd into it and ran ``.\vmlinuz root=./rootfs.img rootwait ro`` (I also tried ``.\vmlinuz root=.\rootfs.img rootwait ro``, just in case)

Results are exactly the same as in my first message.

alsldlflglhljlk, 2024-04-22:
The `root` argument should be what you expect the device to show up as in `/dev`, not a local file. If you are trying to load an initramfs, use `initrd=initrdnamehere.img` instead.

Try renaming `vmlinuz` to `kernel8.img.efi` and doing the same thing you did before, only invoking `.\kernel8.img` (not a typo, don't add the `.efi`) instead. If that works,  the answer is simple – the kernel is not properly configured to support UEFI.

Aegeontis, 2024-04-23:
> The root argument should be what you expect the device to show up as in /dev, not a local file.

I dont quite understand what you mean by this. I have an iso file with a rootfs.img file, a kernel and an initramfs (and some other files like grub). I dont have an unpacked/preinstalled root ready to go on my usb.

Afaik this should really matter though, as even without a rootfs or initramfs, the kernel should be able to at least print something?

> Try renaming vmlinuz to kernel8.img.efi and doing the same thing you did before, only invoking .\kernel8.img (not a typo, don't add the .efi) instead. 

Same results as before.

> If you are trying to load an initramfs, use initrd=initrdnamehere.img instead.

I tried loading an initramfs too, but that did nothing more than adding `EFI stub: Loaded initrd from command line option` to the output.

> If that works, the answer is simple – the kernel is not properly configured to support UEFI.

Could you specify what could be wrong with the kernel? The Fedora CoreOS docs say:
> You can boot the live ISO in either legacy BIOS or UEFI mode, regardless of what mode the OS will use once installed.

[fedora coreos docs](https://docs.fedoraproject.org/en-US/fedora-coreos/bare-metal/#_installing_from_live_iso)